groupgets/pylepton #5: Program gets stuck on garbage frame detection.

Yemen4u2, 2016-07-15:
When I tried running the program it gets stuck on line 156 of Lepton.py and I'm not sure why. It was working a minute ago, but now it just keeps detecting all readings as garbage frames.

junkert, 2016-12-09:
I have the same issue.

junkert, 2016-12-10:
This happens when the SPI device is not responding. Maybe a better error message could be used here.

Yemen4u2, 2016-12-15:
There is no error message. The only way I even detected this was by putting a bunch of `print` statement in my code and finding where the code got stuck.

larryfkm, 2017-01-24:
Is there any way to reset and recover from this situation instead of getting stuck forever here?

hanhanhankim, 2017-06-16:
I have this issue also--any fix yet?

kekiefer, 2017-06-16:
There was some advice on google groups on how to get LEP_RunOemReboot working through python (due to the lack of a python i2c/cci command processor in this project). https://groups.google.com/forum/#!msg/flir-lepton/aUU_F_iswlE/xSil-2XdBAAJ

It would be great to incorporate something like this into pylepton if it could be built easily.

phishstang65, 2018-02-08:
I am getting this same issue. Has anyone tried putting VIN on a Raspberry Pi 3.3V GPIO and toggled the GPIO to reset the Lepton when this condition occurs?

david-ra, 2018-02-28:
The same issue Here. Im using RPI3 with pylepton v3. This is the output of the console:

$ pylepton_capture out.png
Garbage frame number resetting...
Garbage frame number resetting...
Garbage frame number resetting...
Garbage frame number resetting...
Garbage frame number resetting...

I attach a image of the breackboard scorpio v7 ( without mosi pin, i dont know why ):
[image]

david-ra, 2018-02-28:
Problem solved. Just a cable problem in the GPIO :P

rmnd0, 2019-02-27:
> Problem solved. Just a cable problem in the GPIO :P

To wich GPIO pin did you connect CS?

rmnd0, 2019-03-07:
Hardware:
> Rpi zero w.
> FLIR Lepton breakout board.
> FLIR Radiometric Lepton 3.5.

Software:
> Raspbian Stretch.

GPIO pinout:
CS -> Pin 24, CE0
MOSI -> Pin 19, MOSI
MISO -> Pin 21, MISO
CLK -> Pin 23, CLK
GND -> Pin 6, GND
VIN -> Pin 1, 3V3
SDA -> Pin 3, SDA
SCL -> Pin 5, SCL

It doesn't matter if using lepton3 or lepton3-dev,  the output of pylepton_capture is always "Garbage frame number resetting...".

I'm running out of ideas before I declare the thermal core broken.

Any thoughts?

Thank you.

david-ra, 2019-03-10:
I strongly recommend to use the pure thermal 2 module :)

rmnd0, 2019-03-10:
I might try that, thanks ;)

Sent with [ProtonMail](https://protonmail.com) Secure Email.

‐‐‐‐‐‐‐ Original Message ‐‐‐‐‐‐‐

closed 2019-05-16 by kekiefer

TinManAkshay, 2021-02-11:
Hi,

As i have seen people posting here about black images and garbage frame detection.

When i run pylepton_overlay -a 255 example, I get black images. I am using Lepton 3 and breakout out board v2. I have not found the cause of these black images. Any help would be highly appreciated if anyone has fixed the issue so far.

Thanks,
Akshay

AF079, 2022-12-10:
Hi all,
Any solutions for "Program gets stuck on garbage frame detection" and/or the camera just not sending frames after a certain amount of time?

Thanks