groupgets/pylepton #7: Weird images

pranali3215, 2017-01-20:
I am using a FLIR Lepton Camera and a raspberry pi3 to generate heat maps.
In the beginning, the camera gives me the correct output, however, after some time, it gives me following kind of images:
[image]

After I restart the pi, it starts giving me correct images but deteriorates after some time again.
Does anyone have any idea why this can be?
I have an idea that this could be about connections with pi3 as I previously used this with pi1 and it never gave me any such troubles.
Are there any changes in the way lepton camera is connected to pi3 as compared to pi1?

pranali3215, 2017-02-28:
Update:

I thought it was a hardware issue, so I tried it again with a new camera and a new Breakout Board. However, it still threw me same images.

I installed LeptonModule and executed raspberry_video. I ran it for approximately half an hour, and I never got any striped images.

It seems there is a difference in how LeptonModule and pylepton reads images.
Any help?

dirkstoop, 2017-09-13:
I had a similar issue, using pylepton on a Raspberry Pi Zero. My malformed images typically showed a big horizontal band that was black with some white parts. I researched what speeds the SPI port on the Pi Zero supports, and apparently it's at 25MHz and any whole fraction of that. The pylepton code seems to be configured at 18MHz so that shouldn't work.

In Lepton.py, look for this line:
`  SPEED = 18000000`
I changed it to
`  SPEED = 25000000`

and the issue became less very rare.

After that, I changed it to:
`  SPEED = 25000000 / 2`

And I haven't seen a bad frame since. Not entirely sure it's solved, but sure looks like it.

cluelessperson, 2018-03-07:
According to the Lepton datasheet, there's a syncronization procedure and I don't think it's implemented in pylepton at all. 

I think Pylepton is a proof of concept, but isn't meant to be production.

closed 2019-05-16 by kekiefer