groupgets/pylepton #11: Attribute Error: Lepton3 object has no attribute '_msg_size'

jarriaga25, 2017-05-09:
when running pylepton_capture getting above error. I'm also not sure but wondering in which folder I should be running the pylepton_capture command since I see it in a few places. Any assistance is much appreciated, it's been 10yrs since I coded and I'm very rusty and new to linux/python.

kekiefer, 2017-05-09:
It sounds like your system has the old version installed system-wide (via pip, or a previous setuptools install). Try to uninstall that (pip uninstall pylepton, or if that doesn't work, rm -r /usr/lib/python<your version here>/site-packages/pylepton*), then try again. You don't have to install system-wide if you're operating in the root project directory.

jarriaga25, 2017-05-10:
Thanks! That cleared up that issue but now capture just hangs indefinitely. Going to try and get a new pi and start from scratch.

closed 2017-05-10 by kekiefer