groupgets/pylepton #12: Color palettes

alduxvm, 2017-05-17:
Hi!

This is not really an issue, but its a question, is there a way of using color palettes similar to the ones in https://github.com/groupgets/LeptonModule/blob/master/software/raspberrypi_video/Palettes.cpp ??

Thanks!

kekiefer, 2017-05-17:
One way would be to run it through OpenCV's LUT:
http://docs.opencv.org/2.4/modules/core/doc/operations_on_arrays.html#lut

See here for an example:
http://stackoverflow.com/a/37656779

alduxvm, 2017-05-17:
Good point, I forgot about that... I end up using this: `cv2.applyColorMap(gray, cv2.COLORMAP_JET)`

Original: 
[image]

After the applyColorMap:
[image]

Thanks!

closed 2017-05-17 by alduxvm

kekiefer, 2017-05-17:
Sweet!