groupgets/pylepton #15: Using 160x120 sensor

dwallener, 2017-07-14:
I believe these are by definition Lepton 3s? Anyway, the code as-is does work with my sensor. This image is correctly identifiable as a shot of my keyboard:

[image]

But the resolution of the final image is 80x60. And the obvious scanline effect does suggest a single 160px line is being broken into two 80px lines. I've tried the obvious changes in `Lepton.py` - changing the values of the row and column constants - but it returns this:

> Capturing...
Traceback (most recent call last):
  File "./pylepton_capture", line 38, in <module>
    image = capture(flip_v = options.flip_v, device = options.device)
  File "./pylepton_capture", line 10, in capture
    a,_ = l.capture()
  File "/home/pi/pylepton/pylepton/Lepton.py", line 153, in capture
    Lepton.capture_segment(self.__handle, self.__xmit_buf, self.__msg_size, self.__capture_buf[0])
  File "/home/pi/pylepton/pylepton/Lepton.py", line 125, in capture_segment
    ret = ioctl(handle, iow, xs_buf[xs_size * (120 - messages):], True)
IOError: [Errno 90] Message too long

kekiefer, 2017-07-14:
Yes, 160x120 is lepton3 ... you must use the lepton3 or lepton3-dev branch as the master branch does not support lepton 3 yet.

dwallener, 2017-07-14:
Just to confirm that other than the resolution munging, it actually works pretty well...I "descanned" the original image by mogrifying to PPM and manually tweaking the image resolution. It looks good!

[image]

Will look at the alternate branch - thanks!

kekiefer, 2017-07-14:
You've done the right thing. Note that the Lepton 3 transmits four of these 160x30 vospi frames for each frame the camera captures that just need to be glued together to form the 160x120 frame, and you can certainly do this manually. That alternate branch includes the extra buffering to glue them together and some logic to make sure they are captured in order, but it's nothing earth shattering.

closed 2017-07-14 by kekiefer