groupgets/pylepton #19: Jetson TX2 lepton3

HailinRen, 2017-11-11:
Hello! I want to use jetson TX2 and lepton 3. I follow https://devtalk.nvidia.com/default/topic/1024806/jetson-tx2/how-to-enable-spi-spidev-on-28-1-on-target-/ and set the maximum speed as http://me.knnect.com/blog/how-to-connect-flir-lepton-module-to-nvidia-jetson-tk1/. And I also change the default spi device to "/dev/spidev3.0". 
However, when I tried to capture image using "pylepton_capture output.png", I got following error,
"
Traceback (most recent call last):
  File "/usr/local/bin/pylepton_capture", line 37, in <module>
    image = capture(flip_v = options.flip_v, device = options.device)
  File "/usr/local/bin/pylepton_capture", line 9, in capture
    with Lepton(device) as l:
  File "/usr/local/lib/python2.7/dist-packages/pylepton/Lepton.py", line 84, in __enter__
    ioctl(self.__handle, SPI_IOC_RD_MODE, struct.pack("=B", Lepton.MODE))
IOError: [Errno 25] Inappropriate ioctl for device

"

Could anyone help me solve this problem? Thank you!

kekiefer, 2017-11-11:
The second article mentions changing the `MODE` constant to `SPI_MODE_0`, and his device tree contains `spi-cpha;`. Do either of these things help?

HailinRen, 2017-11-11:
Thank you for your note. I have changed these options and now when I run "pylepton_capture output.png" It got stuck. After I applied keyboardInterrupt, it shows following 
Traceback (most recent call last):
  File "/usr/local/bin/pylepton_capture", line 37, in <module>
    image = capture(flip_v = options.flip_v, device = options.device)
  File "/usr/local/bin/pylepton_capture", line 10, in capture
    a,_ = l.capture()
  File "/usr/local/lib/python2.7/dist-packages/pylepton/Lepton.py", line 158, in capture
    time.sleep(0.185)
Do you have any idea? Thank you so much!

kekiefer, 2017-11-12:
Are you on one of the lepton 3 branches?

HailinRen, 2017-11-13:
Thank you for reminding me that. I used lepton 3 branches and got one weird image
[image]
Do you have any idea? I tried both on TX2 and raspberry pi 3 model b (I used pin 17, 19, 21, 23, 25, 24) . if use raspberry pi 3 model b (I used pin 17, 19, 21, 23, 25, 26), it shows garbage frames in terminal. I am also wondering whether my sensor failed in hardware. Actually the image doesn't change when pointing to different directions.

kekiefer, 2017-11-13:
Does it always do that? Certainly, pylepton does not handle the lepton 3 very well. But when doing still capture you should get a good frame at least some of the time.

HailinRen, 2017-11-13:
Yes. It always do that and did not give me any good frame. Do you have any idea?

kekiefer, 2017-11-13:
It looks like an out of sync image, but like I said it shouldn't always be like this. Have a look at https://groups.google.com/forum/#!forum/flir-lepton, there are a handful of different projects that people have done recently to grab data from the Lepton 3, maybe one of those will work better for you.

closed 2019-05-16 by kekiefer