groupgets/pylepton #21: Weird image capture

Goczilla, 2018-01-11:
Hello,

I am discovering this library which seems to be very helpful to acquire data from a Lepton camera.
However, when I try to acquire an image, first the time to execute the capture function of Lepton.py file is very random (between 0.01 and 30s), and second the image display is weird. I can see my "shape" but with a lot of noise or like if pixels are dead.
But, using an old software of my colleague (using C++/Python and Perl), the camera is ok and works correctly.
So it deals with my code but I can see where.

Can you help me please ? I thank you in advance.
```
import sys
import numpy as np
from pylepton import Lepton
import matplotlib.pyplot as plt

with Lepton() as l:
	image = np.zeros((60, 80, 1), 'float')  # to perform operation after
	
	# Capture image
	_, new_sum = l.capture(image)

	# Normalize between 0 and 1
	data = np.reshape(np.rot90(image,1),60*80,1)
	data2=data-min(data)
	data2=data2/max(data2)
	data2=np.reshape(data2, (60,80))
	data = 255.*data2
	data = np.squeeze(image)

# Plotting image acquired
fig = plt.figure()
im = plt.imshow(data)
pos = fig.add_axes([0.93,0.05,0.05,0.8])
fig.colorbar(im, cax=pos) 
plt.show()
```

kekiefer, 2018-01-11:
It would be best to post this question on the [Lepton Google group](https://groups.google.com/forum/#!forum/flir-lepton) along with an image of the artifact you're seeing (this will really help), and I'll be happy to review your code.

closed 2018-01-11 by kekiefer

Goczilla, 2018-01-12:
[Done](https://groups.google.com/forum/#!topic/flir-lepton/Due4UHJgLAc)

Goczilla, 2018-01-26:
Hello,

Sorry for disturbing, but can you look at my post on the google group?

Thank you in advance ;)