groupgets/pylepton #22: Inappropriate ioctl for device

tictag, 2018-01-21:
I've tried running a basic capture per the README.MD instructions under both Python v2.7.9 (with Numpy v1.8.2 & OpenCV v2.4.9.1) and Python 3.4.2 (with Numpy v1.8.2 & OpenCV v3.4.0) and always receive the following errors:
**Python v2.7.9**
```
Python 2.7.9 (default, Sep 17 2016, 20:26:04)
[GCC 4.9.2] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> from pylepton import Lepton
>>> with Lepton() as l:
...   a,_ = l.capture()
...
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python2.7/dist-packages/pylepton/Lepton.py", line 84, in __enter__
    ioctl(self.__handle, SPI_IOC_RD_MODE, struct.pack("=B", Lepton.MODE))
IOError: [Errno 25] Inappropriate ioctl for device
```
**Python v3.4.2**
```
Python 3.4.2 (default, Oct 19 2014, 13:31:11)
[GCC 4.9.1] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from pylepton import Lepton
>>> with Lepton() as l:
...   a,_ = l.capture()
...
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.4/dist-packages/pylepton/Lepton.py", line 84, in __enter__
    ioctl(self.__handle, SPI_IOC_RD_MODE, struct.pack("=B", Lepton.MODE))
OSError: [Errno 25] Inappropriate ioctl for device
```
Doing a `ls /dev` reports that my SPI device is `spidev0.0`.

I'm a complete newbie to Linux and Python and I just have no idea where to even start troubleshooting this, frankly I don't even know what the error means!

Any help very much appreciated :)

junkert, 2018-01-24:
Try something like this:

```
def capture(flip_v=False, device="/dev/spidev0.1"):
    with Lepton(device) as l:
        data, _ = l.capture()
```

kekiefer, 2019-05-16:
What was the problem here? Can we close this issue?

amosgolan1, 2019-10-29:
Hi, I'm getting a similar error and can't resolve by substituting the device to device="/dev/spidev0.1. was this ever resolved? I don't really understand what my error means as well...

I ran the example usage code (saved the file on my machine as "test_pylepton.py") and tried to run it using python3 and got this error:

Traceback (most recent call last):
  File "test_pylepton.py", line 5, in <module>
    with Lepton() as l:
  File "/Users/.../GetThermal/pylepton/pylepton/Lepton.py", line 82, in __enter__
    self.__handle = open(self.__spi_dev, "wb+", buffering=0)
PermissionError: [Errno 1] Operation not permitted: '/dev/spidev0.0'

Could you please advise what I'm doing wrong here?
My camera and setup work fine with GetThermal or other apps.

Any help would be very much appreciated 
Thanks!

kekiefer, 2019-10-29:
GetThermal? Do you have a breakout board or a PureThermal board?

amosgolan1, 2019-10-29:
Thanks for your reply @kekiefer. I have a PureThermal Mini connected to my mac over USB.
https://store.groupgets.com/products/purethermal-mini-flir-lepton-smart-i-o-module

I meant that I was able to follow the installation of the GetThermal app:
https://github.com/groupgets/GetThermal

And it works well with my camera, as well as just using it as a "normal" webcam with OBS. I'm interested in using the PureThremal mini with OpenCV+python to do some analysis and get temperature readings for each pixel. I was hoping to use pylepton to do so.
Thank you for your help!

kekiefer, 2019-10-29:
pylepton is just for the breakout board. You should look at https://github.com/groupgets/purethermal1-uvc-capture

closed 2019-10-29 by kekiefer

al2k, 2021-05-23:
Check your cable setup. An incorrect setup will throw this error when using pylepton 3.0. 

This guide from FLIR shows the correct GPIO setup: https://lepton.flir.com/forums/topic/pylepton-overlay-guide-2-0-for-raspbian-pixel/