groupgets/pylepton #23: Jetson TK1 Lepton 2.5

izackthegreat, 2018-03-31:
I was following http://me.knnect.com/blog/how-to-connect-flir-lepton-module-to-nvidia-jetson-tk1/ to work on getting my Lepton working with my Jetson. Whenever I run the pylepton_capture, the program seems to get get caught somewhere. Whenever I stop the program with a keyboard interupt I get the following traceback 

File "pylepton_capture", line 37, in <module>
    image = capture(flip_v = options.flip_v, device = options.device) 
File "pylepton_capture", line 10, in capture
     a, _ = l.capture()
File "pylepton/Lepton.py", line 158, in capture
     time.sleep(0.185)

Any help would be appreciated.

junkert, 2019-03-01:
This usually means the sensor is wired incorrectly or the sensor is broken. I recommend wiring it up to a oscilloscope or digital-analyzer to make sure the Lepton is still communicating over the SPI bus.

kekiefer, 2019-05-16:
Can this issue be closed?

Henrikcox, 2019-11-07:
I have the same issue and just tried checking the sensor's output with An Analog Discovery 2. When plugged into the Lepton's MOSI pin, I do get data streaming so it doesn't seem to be broken. Are there other possible explanations?