groupgets/pylepton #26: Lenpton3_dev can't work on python3

qingshutie, 2018-08-30:
I install with sudo python3 setup.py install . but  import pylepton.Lepton3  error, ImportError : No module named 'ioctl_numbers'

kekiefer, 2019-05-16:
Did you resolve this? It's not a general python3 issue since this module has been update to work with python3 more than a year before this was posted.

briandinh2307, 2020-08-14:
[SOLVED]

The problem is in the file lepton3.py. 

Change line 8 and 11 in Lepton3.py by adding a "." and include bracket "()" in all the print statements should fix the problem
ln 8: from ioctl_numbers import _IOR, _IOW
--> from .ioctl_numbers import _IOR, _IOW
ln 11: from Lepton import Lepton
--> from .Lepton import Lepton