groupgets/pylepton #27: Reliable vospi sync via the breakout board

gusisin, 2018-10-04:
Hi everyone,
I came across this while browsing github. Basically on the back of the breakout board is a GPIO port labelled GPIO3. This provides a sync signal that can be connected to a PI gpio port to provide bullet proof sync between the camera and the break out board. Dan Julio produced this project (see below)

Dan Julio's project

leptonic with breakout board vsync

My question is, would it be possible to include a wired vsync as an option to pylepton ? Has anyone looked at this before?

Gus

kekiefer, 2019-05-16:
It should be possible. But first in order to do this, you'd have to add an I2C interface to pylepton, to change enable this configuration on the Lepton. Pull requests welcome.

TinManAkshay, 2020-07-13:
@gusisin Did you try enabling the vsync pin by adding the I2C interface? If Yes, did you make changes inside the pylepton library or the main code where you using the pylepton library?

Thanks,
Akshay