groupgets/pylepton #29: PyLepton Not working on raspberry pi

GitHubVik, 2019-03-01:
I have downloaded and installed the pylepton from github and wired it up to my raspberry pi, but when the pylepton capture code is run, it runs indefinitely without outputting the image, and when I force it to quit it says it was at a time.sleep() command. Any ideas?

Cullpepper, 2019-03-01:
Start with the easy stuff?

   - double check the wiring to the Lepton is on the right pins
   - double check the lepton lens is firmly seated into the lepton board

GitHubVik, 2019-03-01:
I have checked the wiring, with the clk in both ce1 and ce0, but to no avail. The lepton works on an old repo called lepton module master, which was on the pi before I started working on this project, but I can't get it to work in python.

junkert, 2019-03-01:
CE0 and CE1 most definitely should not be wired to the SPI-CLK pin.

junkert, 2019-03-01:
I am using this codebase successfully (on master) with RPi2 & 3 with both a Lepton 2.5 and 3.5 with no problems. Every time I have an issue it is usually associated with wiring, incorrect configuration, or a broken sensor.

GitHubVik, 2019-03-01:
I got it to run, but now when I display the image that it saved it shows a black/grey screen even when I put my hand or something hot in front of the camera

junkert, 2019-03-01:
This sounds like an implementation issue and not the library itself. Without seeing your capture code I am unable to help you troubleshoot this.

GitHubVik, 2019-03-01:
I am using the stock pylepton_capture code from this library, I will update this thread after I troubleshoot more tomorrow

junkert, 2019-03-04:
Are you still having issues with the pylepton_capture software? Also have you tried switching the selected CE port in the code? Also make sure you have a pull-down/pull-up resistor on the CE line. You can set this on the Raspberry pi itself -> https://issuu.com/themagpi/docs/the_magpi_issue_2_final
See pages 10-14

Also does your original code that works use the same libraries to communicate with the Lepton?

GitHubVik, 2019-03-04:
I have not tried this pull-down/up resistor that you are talking about. I will try to switch the CE port in the code, for now I have switched to the purethermal mini over usb because that seems to be working. I believe the original code doesn't use python, but the leptonModule library on the flir website.

Cullpepper, 2019-03-05:
Sorry if this is a stupid comment but...

You're aware that the lepton raw output is in greyscale? You have to apply
a colorscale to the image if you want it to look like a commercial Flir
device. If I remember correctly, pylepton_capture doesn't apply any color
to the output all by itself.

That said, if it's working, you should see edges and shapes where the
thermal value is different.

GitHubVik, 2019-03-05:
Yup, I do know its greyscale, but I get no discernable picture when I use the capture code, just a flat grey screen with a white line at the bottom, even when I or something hot is in the frame.

kekiefer, 2019-03-05:
A typical scene will have a very low dynamic range in 14 bits of data, and that needs to be contrast extended if you want to see something. If you have a line at the bottom of the screen that's outside the range of your signal, it's going to cause problems with that contrast extension.

junkert, 2019-03-06:
As @kekiefer said, make sure to use a hot object in front of the camera. You can also use OpenCV to recolor the image and normalize it as well. Check some of my code here for examples -> https://github.com/sentient-controls/plantspy/blob/master/plantspy.py#L105

closed 2019-05-16 by kekiefer

briandinh2307, 2020-08-13:
I'm trying to use python3 with pylepton. When I run my script. it said No module named 'Lepton'. I need to use python3 for my project. Does anyone know how to solve this issue.

Thanks