groupgets/pylepton #30: Not working with Pi 3b+ and Lepton3.5

tannerliu347, 2019-03-14:
I tried using master, lepton3 branch and lepton3-dev branch. I can get an output image by using master branch, but the image looks distorted.
[image]
With lepton3 and lepton3-dev branch, I kept getting Lepton3 object has no attribute '_msg_size' error.
Traceback (most recent call last):
  File "/usr/local/bin/pylepton_capture", line 37, in <module>
    image = capture(flip_v = options.flip_v, device = options.device)
  File "/usr/local/bin/pylepton_capture", line 9, in capture
    with Lepton3(device) as l:
  File "/usr/local/lib/python2.7/dist-packages/pylepton/Lepton3.py", line 32, in __init__
    self._xmit_buf = np.zeros((self._msg_size * Lepton.ROWS * 4), dtype=np.uint8)
AttributeError: 'Lepton3' object has no attribute '_msg_size'

Any thoughts will be appreciated. Thanks

eKawlan, 2019-04-01:
I have the same issues.
It seems, no one able to help!

eKawlan, 2019-04-01:
Hi tannerliu347,
If have anything new, please let me know.
I do the same.

Thanks

tannerliu347, 2019-04-01:
@eKawlan Hello ekawlan. I had luck using the lepton 3-dev branch. The video streaming still loses frame frequently, but the capture function works.

kekiefer, 2019-05-16:
Agree, this is what it looks like when you're using Lepton 2 code to read from a Lepton 3.

closed 2019-05-16 by kekiefer

TinManAkshay, 2022-05-24:
Hello @tannerliu347  - 

Did you happen to get L:epton 3/3.5 up and running using lepton-3 dev? In other words, you are getting full 160*120 packets out without any issues?

Thanks,
akshay