groupgets/pylepton #31: Unable to detect the camera

Pooja-Baviskar, 2019-03-19:
I am getting the following error after connecting lepton camera with Rasberry Pi 3 B

mmal: mmal_vc_component_create: failed to create component ‘vc.ril.camera’ (1:ENOMEM)

mmal: mmal_component_create_core: could not create component ‘vc.ril.camera’ (1)

Traceback (most recent call last):

File “/usr/local/bin/pylepton_overlay”, line 66, in 

main(flip_v = options.flip_v, alpha = options.alpha, device = options.device)

File “/usr/local/bin/pylepton_overlay”, line 17, in main

with picamera.PiCamera() as camera:

File “/usr/lib/python2.7/dist-packages/picamera/camera.py”, line 431, in __init__

self._init_camera(camera_num, stereo_mode, stereo_decimate)

File “/usr/lib/python2.7/dist-packages/picamera/camera.py”, line 460, in _init_camera

“Camera is not enabled. Try running ‘sudo raspi-config’ “

picamera.exc.PiCameraError: Camera is not enabled. Try running ‘sudo raspi-config’ and ensure that the camera has been enabled.

But I have enabled the camera and checked all the connections as well

junkert, 2019-03-19:
pylepton_overlay uses a regular CCD camera (RaspberyPI camera) + the Lepton to create a LWIR Image + visible CCD image.

Try using the pylepton_capture instead.

kekiefer, 2019-05-16:
Has this been resolved? The `ENOMEM` error trying to open the camera is what looks fishy to me.