groupgets/pylepton #33: Is it possible to Stop Camera from adjusting exposure

BespinV, 2019-05-16:
I am trying to lock the scale/heat range of the camera so that it doesn't adjust based on the scene it is taking a photo of. 

For example, when my hand is in the frame, the resulting image that the camera takes only highlights my hand (as it is the hottest thing in the frame). However, when I remove my hand, it auto compensates and adjusts the heat range to now show the next hottest thing as a bright pixel (so now the whole background is lit up with white pixels.)

Is it possible to lock the auto adjust feature of the camera? Let me know you need any clarification for my explanation. I apologize if it is confusing...

junkert, 2019-05-16:
I recommend looking into the I2C interface and setting these settings there -> https://www.flir.com/globalassets/imported-assets/document/flir-lepton-software-interface-description-document.pdf

I'm not sure if anyone has added the ability to change these settings in this library, but it would be pretty easy to add I2C control as well.

kekiefer, 2019-05-16:
Autoranging in the pylepton examples has nothing to do with Lepton settings since pylepton captures in raw mode. It actually has nothing to do with pylepton at all, just the examples.

The example capture program and pasting the example code have calls cv2.normalize, which is extending the contrast. Replace that with something suiting your application.

Note that if you don't have a radiometric Lepton, the center of your range will float around with the temperature of the camera.

closed 2019-05-16 by kekiefer