groupgets/pylepton #34: IOError: [Errno 25] Inappropriate ioctl for device

cjcbusatto, 2019-07-18:
I am experiencing the issue running the example code from the README.md with a Lepton3.5 on Ubuntu 18

```bash
$ uname --all
Linux ubuntu 4.18.0-25-generic #26~18.04.1-Ubuntu SMP Thu Jun 27 07:28:31 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux
```

```bash
$ ls /dev/
$ ...
$ -rw-r--r--  1 root  root             0 Jul 18 17:42 spidev0.0
$ -rw-r--r--  1 root  root             0 Jul 18 17:41 spidev0.1
$ ...
```

Code:

```python
import numpy as np
import cv2
from pylepton import Lepton

with Lepton() as l:
  a,_ = l.capture()
cv2.normalize(a, a, 0, 65535, cv2.NORM_MINMAX) # extend contrast
np.right_shift(a, 8, a) # fit data into 8 bits
cv2.imwrite("output.jpg", np.uint8(a)) # write it!
```

Output:

```bash
cbusa@ubuntu:~/pylepton$ sudo python main.py 
Traceback (most recent call last):
  File "main.py", line 5, in <module>
    with Lepton() as l:
  File "/home/cbusa/pylepton/pylepton/Lepton.py", line 84, in __enter__
    ioctl(self._handle, SPI_IOC_RD_MODE, struct.pack("=B", Lepton.MODE))
IOError: [Errno 25] Inappropriate ioctl for device
```

What I have already tried:
- Change from Lepton() to Lepton("/dev/spidev-0.1")
- Use different branches of the project
- Tested with the `pylepton_capture` application
All of them produce the same error.

Interesting fact is: when I use the `GetThermal-v0.1.4-x86_64-Linux.AppImage`, I have no issues. 

Do you have any idea how should I proceed with the analysis?

kekiefer, 2019-07-18:
pylepton doesn't work with the PureThermal boards, it is only for the "classic" breakout.

If you're interested in using python to hook into the camera, have a look at this repository for examples: https://github.com/groupgets/purethermal1-uvc-capture

closed 2019-07-18 by kekiefer

cjcbusatto, 2019-07-18:
Thank you, with the mentioned repository it worked flawless!