groupgets/pylepton #37: Hanging terminal - pylepton_capture output.png

Henrikcox, 2019-11-26:
I am working with a raspberry pi 3 (and have also tried on a raspi zero w) and the [FLIR radiometric dev kit from SparkFun](https://www.sparkfun.com/products/14654) (Lepton 2.5). 

When I run 
`pylepton_capture output.png`

I get an unresponsive terminal, after hitting **Ctrl-C** I get the following output

`File "pylepton_capture", line 37, in
image = capture(flip_v = options.flip_v, device = options.device)
File "pylepton_capture", line 10, in capture
a, _ = l.capture()
File "pylepton/Lepton.py", line 158, in capture
time.sleep(0.185)`

I have tested the Lepton on a digital analyzer and the CS and MISO channels of the SPI seem to be communicating OK. SCK is identical to the CS channel though and I am unsure of the cause or fix to this.

Digging into the code, it seems to get stuck in the `def capture` function (in Lepton.py) in the `while True` loop. It seems that the `capture_buf` array is never populated with data from the pixels collected by the Lepton. 

I'm not 100% sure this is the root of the issue, or what this affects either. Any advice on this issue in general is appreciated in advance.

superbuck, 2020-02-18:
I had the same problem, turned out that it was wrong wiring problem.

alexandre-mazel, 2020-05-04:
And so for more information, the documentation say to connect CS to 26 but it works on my Raspberry 3 when I connect to 24 (aka CEO instead of CE1).

Can save you hours of research and doubts...