groupgets/pylepton #38: pylepton_overlay -a 155 error

basit434, 2019-12-25:
When i run "pylepton_overlay -a 155" command i get the following error. However both my cameras (pi camera and flir lepton are working fine if i run them individually)
When i run "pylepton_capture output.png" this command it also works fine and captures a picture.

i am only facing the issue while running "pylepton_overlay -a 155"

Traceback (most recent call last):
  File "_ctypes/callbacks.c", line 315, in 'calling callback function'
  File "/home/pi/.local/lib/python2.7/site-packages/picamera/mmalobj.py", line 1227, in wrapper
    self._pool.send_buffer(block=False)
  File "/home/pi/.local/lib/python2.7/site-packages/picamera/mmalobj.py", line 1931, in send_buffer
    super(MMALPortPool, self).send_buffer(port, block, timeout)
  File "/home/pi/.local/lib/python2.7/site-packages/picamera/mmalobj.py", line 1881, in send_buffer
    raise PiCameraMMALError(mmal.MMAL_EAGAIN, 'no buffers available')
picamera.exc.PiCameraMMALError: no buffers available: Resource temporarily unavailable; try again later

kekiefer, 2019-12-26:
Your traceback is in the picamera library, not this one. Can you confirm if picamera is working on it's own? https://picamera.readthedocs.io/en/release-1.13/recipes1.html

basit434, 2019-12-26:
yeah the pi camera is working on its own

infact both my cameras are working individually at same time
the lepton camera is working through lepton module master

basit434, 2019-12-26:
kindly help me this is my final year project and i am left with only one
week

kekiefer, 2019-12-26:
I'm not asking if the camera is working, you already said that it does. I'm asking specifically if the `picamera` library is working, that this project relies on. I've shared a link to examples for this library.

basit434, 2019-12-26:
yeah its working

basit434, 2019-12-26:
???

kekiefer, 2019-12-26:
OK, given that I've pointed you to the fact that this is a problem the picamera library, did you try searching for related issues there?

https://github.com/waveform80/picamera/issues?utf8=✓&q=is%3Aissue+is%3Aopen+no+buffers+

There are a number of threads with the same error. This one suggests an alternative to using the camera overlay function:

https://github.com/waveform80/picamera/issues/448

Given your time crunch, you should also be looking at other ways you can combine the picamera image and the lepton image. The example given here is just that, an example, and not the only way to get the job done.

basit434, 2019-12-26:
> waveform80/picamera#448

all the examples on
https://picamera.readthedocs.io/en/release-1.13/recipes1.html
are working fine

hrdunn, 2020-05-11:
@basit434 did this ever get resolved? I am running into the same issue, the picamera lib works fine on its own.

basit434, 2020-05-11:
yeah

hrdunn, 2020-05-14:
@basit434 what was the solution?

basit434, 2020-05-14:
using the alpha transparency it worked
i was able to merge both the videos

Ailsamly, 2020-10-16:
hi, I have the same problem, how did you solve it? @basit434 @hrdunn

hrdunn, 2020-10-16:
@Ailsamly yes I used a different library, this lib is garbage, out of date, and borderline deprecated, I managed to get it to work with a lepton 3.5 using the `lepton-3-dev` branch. I made a few mods and I think I commented out the overlay. If you have a breakout board take a look at [Lepton Module](https://github.com/groupgets/LeptonModule) a C lib, but don't let that scare you, I never wrote C before and got it working or if you have a purethermal board (so much easier to work with) take a look at [purethermal1-uvc-capture](https://github.com/groupgets/purethermal1-uvc-capture) a python library that you can extend

Ailsamly, 2020-10-16:
> @Ailsamly yes I used a different library, this lib is garbage, out of date, and borderline deprecated, I managed to get it to work with a lepton 3.5 using the `lepton3-dev` branch. I made a few mods [...]

@hrdunn Thanks very much for your quick and kind reply! But I am really new to this as well as programming.. could you please specify more about your steps?  I am also using a lepton 3.5 with the latest breakout board. Thank you!!!

hrdunn, 2020-10-16:
@Ailsamly sorry dude I wish I could, I deleted the modified library and ditched the breakout board back shortly after getting it to work in May so I don't have my steps that got it to work.

basit434, 2020-10-16:
its ok

Ailsamly, 2020-10-16:
> its ok
> […](#)

@basit434 how did you solve this?