groupgets/pylepton #39: Lepton 2.5 and 3.5 are not working with New Raspbian release: Buster

TinManAkshay, 2020-02-11:
Hi,

Earlier I used these two FLIR cameras on Raspberry pi 3 with Raspbian Stretch, its working absolutely fine. But I want to try them with Raspberry 4 with the Raspbian current release: Buster. It's not working though. Are these cameras and their libraries are compatible with new Raspbian release on Pi 4?

Thanks!

kekiefer, 2020-02-11:
Newer raspbian has a frequency scaling "feature" that can cause problems since it also adjusts the SPI clock. This functionality can be disabled. See here: https://github.com/groupgets/LeptonModule/tree/master/software/raspberrypi_video#lepton-3x-1

TinManAkshay, 2020-02-11:
I am running everything on boot-up of the Pi. Can you tell me the way how can I set that command line so that it can work on boot?

kekiefer, 2020-02-11:
Well does the change work?

How about this https://askubuntu.com/questions/929884/how-to-set-performance-instead-of-powersave-as-default

Otherwise I would just open this file from your python program and write to it.

TinManAkshay, 2020-02-12:
That didn't work. All I can see the Red LED is on and Green LED starts flashing (since you know it starts booting up the sd card) when i give the power supply. I have a .py code which is in /usr/bin which is not running during start-up with Raspbian buster on Pi 4 but when I run the same thing with Raspbian Stretch on Pi 3 B+, it works. Can you give an idea where I am wrong with Pi 4?

kekiefer, 2020-02-12:
I'm afraid I don't understand. Do you have a specific error or problem with pylepton that I can help you with?

TinManAkshay, 2020-02-12:
With HDMI output, when I execute the .py code on terminal, it says error with /boot/thermal_cal.json. Is it related to pylepton libraries?

kekiefer, 2020-02-12:
I have no idea what that file is or what would read it. This doesn't sound related to pylepton.

TinManAkshay, 2020-02-17:
Hi,

If I put this line in /etc/rc.local .......sudo sh -c "echo performance > /sys/devices/system/cpu/cpufreq/policy0/scaling_governor"......... 

Will this change work during startup?

junkert, 2020-02-19:
It should, but you really should look into using a more permanent solution the Debian way:
https://wiki.debian.org/HowTo/CpuFrequencyScaling

Taken from StackOverflow (https://raspberrypi.stackexchange.com/questions/9034/how-to-change-the-default-governor):
```
Since Raspbian is based on Debian, you should be able to find the packages cpufrequtils and sysfsutils. After installing them, you can set the governor and other parameters from /etc/default/cpufrequtils and /etc/sysfs.conf
```

TinManAkshay, 2020-02-19:
I understood your point but when i make these changes thermal sensor works but other things like distance sensors and accelerometer (which I have connected to Pi 4) do not work according to the raspi-config, governor = "ondemand". So these conflicting governor configurations are messing up with readings from thermal sensor, distance sensors and accelerometer. 
In nutshell, when I get readings from thermal sensors then I don't get readings from other things or vice-versa. So how can I make changes so that everything run on governor="performance"?

Thanks!

closed 2021-02-25 by TinManAkshay