groupgets/pylepton #41: Getting split images with Lepton 3 and FLIR Lepton Breakout Board V2.0

freda997, 2020-04-17:
Hi,

I've been getting the program running using Lepton 3 with the new [release](https://www.flir.com/products/lepton-breakout-board-v2.0/) of the breakout board, but images I am getting are either corrupted or split like the picture below. I tried to adjust the SPEED in Lepton.py and haven't gotten the issues fixed completely, but it does make a difference. So I suspect it is a sync issue with SPI. According to the datasheet, the new version of breakout added a vsync output signal, yet the pylepton library doesn't reflect the changes.  So I wonder could this be the cause of the problem. BTW I am using Raspberry Pi 4. 
[image]

hrdunn, 2020-05-05:
@freda997 hey I'm actually rigging up a lepton 3 as well with a breakout board v2.0 on a rPi 4 I'm still a bit new to all of this, but was wondering which resource you used for setting up your wiring. I was following a guide on FLIR and referencing the datasheet but there aren't any great ones yet for the 2.0 board. Not sure if you'd be willing to provide how you did it? I will let you know if I run into the split frame issue as well, more than happy to colab.

freda997, 2020-05-09:
> @freda997 hey I'm actually rigging up a lepton 3 as well with a breakout board v2.0 on a rPi 4 I'm still a bit new to all of this, but was wondering which resource you used for setting up your [...]

Hi @hrdunn, sorry for not getting back to you in time. Here is the [guide](https://lepton.flir.com/getting-started/raspberry-pi-lepton/) I used when I set up the wiring, and here is the [datasheet](https://flir.netx.net/file/asset/21091/original) for 2.0 breakout board. Let me know if you have any more questions. I am definitely open to collab!

hrdunn, 2020-05-09:
@freda997 I actually managed to get it working using a combo of those things as well. Never wired a board before so I was pretty lost the other day. I am also not experiencing the splitting issue you ran into. I am using the `lepton3-dev` branch though so I think that could be the solution for you if you are still getting the split. I am running into an issue extracting the raw radiometric data. My understanding is there should be some way to get the temperature of each pixel but I have no idea how to do that and it doesn't seem to be documented here. No idea if you have tried that yet. I've got a friend who is an electrical engineer and C++ guy taking a look at this [document](https://groupgets-files.s3.amazonaws.com/lepton/Lepton-3-Engineering-Datasheet.pdf) so if I figure that out I will let you know.

dr-m-nasr, 2020-06-08:
@hrdunn 
did you figure it out?

kekiefer, 2020-07-06:
This is what the output looks like if you're not using the `lepton3` or `lepton3-dev` branch of this code, but are using a Lepton 3 part.

Ailsamly, 2020-10-14:
> @freda997 I actually managed to get it working using a combo of those things as well. Never wired a board before so I was pretty lost the other day. I am also not experiencing the splitting issue [...]

@hrdunn Hi, how is it going? I am new to this and really want to listen to your experience☺️

Ailsamly, 2020-10-14:
> This is what the output looks like if you're not using the `lepton3` or `lepton3-dev` branch of this code, but are using a Lepton 3 part.
Hi Kurt @kekiefer, sorry for contacting you directly here. I saw you replied to this lepton issue and guess you might have a lot of experience with this.  I am really struggling with using Pi zero W, lepton 3.5, and breakout board V2.0. I tried to follow this guide https://learn.sparkfun.com/tutorials/flir-lepton-hookup-guide , but after I installed the example software (https://github.com/groupgets/LeptonModule/tree/master/software/raspberrypi_video)
and run the raspberry pi_video, I didn’t get anything. Here is what showed on the terminal. I would be super appreciated if you could give me some help. Thanks very much!

Best,
Leyuan 

sudo ./raspberrypi_video -tl 3
X Error: BadAccess (attempt to access private resource denied) 10
  Extension:    129 (MIT-SHM)
  Minor opcode: 1 (X_ShmAttach)
  Resource id:  0x1600001
X Error: BadShmSeg (invalid shared segment parameter) 128
  Extension:    129 (MIT-SHM)
  Minor opcode: 5 (X_ShmCreatePixmap)
  Resource id:  0x160000a
X Error: BadDrawable (invalid Pixmap or Window parameter) 9
  Major opcode: 62 (X_CopyArea)
  Resource id:  0x160000b
X Error: BadDrawable (invalid Pixmap or Window parameter) 9
  Major opcode: 62 (X_CopyArea)
  Resource id:  0x160000b
X Error: BadDrawable (invalid Pixmap or Window parameter) 9
  Major opcode: 62 (X_CopyArea)
  Resource id:  0x160000b
X Error: BadDrawable (invalid Pixmap or Window parameter) 9
  Major opcode: 62 (X_CopyArea)
  Resource id:  0x160000b
X Error: BadDrawable (invalid Pixmap or Window parameter) 9
  Major opcode: 62 (X_CopyArea)
  Resource id:  0x160000b
X Error: BadDrawable (invalid Pixmap or Window parameter) 9
  Major opcode: 62 (X_CopyArea)
  Resource id:  0x160000b
X Error: BadDrawable (invalid Pixmap or Window parameter) 9
  Major opcode: 62 (X_CopyArea)
  Resource id:  0x160000b
X Error: BadDrawable (invalid Pixmap or Window parameter) 9
  Major opcode: 62 (X_CopyArea)
  Resource id:  0x160000b
X Error: BadDrawable (invalid Pixmap or Window parameter) 9
  Major opcode: 62 (X_CopyArea)
  Resource id:  0x160000b
X Error: BadDrawable (invalid Pixmap or Window parameter) 9
  Major opcode: 62 (X_CopyArea)
  Resource id:  0x160000b
X Error: BadDrawable (invalid Pixmap or Window parameter) 9
  Major opcode: 62 (X_CopyArea)
  Resource id:  0x160000b
X Error: BadDrawable (invalid Pixmap or Window parameter) 9
  Major opcode: 62 (X_CopyArea)
  Resource id:  0x160000b
X Error: BadDrawable (invalid Pixmap or Window parameter) 9
  Major opcode: 62 (X_CopyArea)
  Resource id:  0x160000b

ahmed-mans, 2021-03-01:
Hi,

I am trying to configure my breakout board v2 with this tutorial : https://lepton.flir.com/getting-started/raspberry-pi-lepton/ .
But I am stuck at the step where it is asked to "Copy the RaspberryPi onto the Pi". I don't really understand what that means. Can anyone help me with that ? I have a RPi 4B, and a lepton 3.5. 

Thanks :)