groupgets/pylepton #43: SPI speed

zoltanszalontay, 2020-05-23:
Is there any way to set the speed of the SPI bus? 

I am using lepton3-dev with an RPI 3B+ and a 4B with a Lepton 3.5 via a Breakout Board v2. I set the CPU governor to run at fixed 100%. Video is fine, but there are sporadic lost frames ("Garbage..." message). Also, when I start picamera, the pylepton almost stops. As soon as I stop picamera, Lepton capture gets smooth again. (OK, that may be because the two I/O operations are blocking each other so threading should be used.)

I suspect the SPI bus speed is slow by default because when I used flirpy with my PureThermal, I could get stable video without any errors only at 30MHz. Anything  below that caused the frames split.

I am aware that pylepton does not use I2C, but I hope basic SPI settings, like bus speed can be configured.

zoltanszalontay, 2020-05-23:
I did not debug it yet, but anyone knows if I can set SPI speed here?
https://github.com/groupgets/pylepton/blob/c856c6a1d44970d878e1f4bfdae2c8eee67a6286/pylepton/Lepton.py#L23

arushk1, 2020-07-05:
Hi @zoltanszalontay I'm facing the same issue. You can set the SPI bus speed at Line 46 in Lepton.py :

SPEED = 18000000

Setting the speed does not do anything to smooth out the video. It's bad when using the PiCamera simultaneously.