groupgets/pylepton #44: Lepton 2.5 and its VSYNC pin

TinManAkshay, 2020-07-06:
Hi everyone,

Months back I was using Lepton 2.0 with breakout board v1.0(which is totally obsolete now). New Lepton 2.5 is coming along with a breakout board v2.0 and this board has introduced a new pin called VSYNC (which was not in the earlier version of breakout board v1.0). 
So I wanted to ask if anybody here got the smooth output from Lepton 2.5 with breakout board v2.0 with its VSYNC pin connected to the Raspberry  pi 3 B+. Did you notice anything wrong with the thermal output in case you don't connect the breakout board's VSYNC pin to the Pi and later may be resolved with VSYNC pin?

Thanks,
Akshay

kekiefer, 2020-07-06:
None of the Linux code available (at least none I am aware of) takes advantage of this feature. In order to enable it, some I2C commands would need to be sent to the Lepton. Pylepton does not support the I2C interface for the Lepton, but it would benefit from this.

TinManAkshay, 2020-07-06:
What if we don't connect the board's VSYNC pin to the raspberry pi? Will not the output be wrong without that?

kekiefer, 2020-07-06:
The pin does nothing unless it's enabled. It's not enabled by the code, and therefore not expected to do anything. It can be connected or not, it doesn't matter.

TinManAkshay, 2020-07-06:
I understand your point. Then what good this pin would bring to the thermal output if its enabled along with i2c pins?

kekiefer, 2020-07-06:
The software wouldn't have to poll to see if there's a new frame available. This is an unreliable process, and causes frequent synchronization loss. It would be able to start downloading each frame exactly as soon as the frame is available.

TinManAkshay, 2020-07-06:
So what i understand from your points so far is that without VYSNC pin, the thermal output frame could face a frequent synchronization loss. To avoid that, we must enable the VSYNC pin to the raspberry pi, which gives the capability of getting the frames as soon as they are available? 

That's what you have been trying to imply right?

TinManAkshay, 2020-07-13:
@kekiefer I haven't heard from you after my previous reply. 

Just wanted to check that if VSYNC is not enabled, we might loose few frames and if this pin is enabled by adding I2C interface from outside, there will be synchronization, as in, no frame loss?

Thanks,
Akshay