groupgets/pylepton #48: what does capture() returns?

wenbozzz, 2020-08-15:
it says 
>Image data from `capture()` is 12-bit, non-normalized (raw sensor data). 

I have a [lepton 2.5 dev kit ](https://www.sparkfun.com/products/15948)
I would assume it should be the temperature value for each pixel
but when I put an melting ice cube (arround 0 degree C) and a cup of coffee (tested with probe, arround 55 degree  C) in front of the sensor, the minimum in the array is `27415` and maximum value is `32709` with doesn't really make sense to me

finesome, 2020-08-26:
The values are actually in centikelvin, which is (K * 100).
So if you have a value of 27415, it means that the temperature value is 274.15 Kelvin, which is 1 Celsius.

wenbozzz, 2020-08-27:
Oh thank you so much for this. Never would have thought it's kelvin 🤣

closed 2020-08-27 by wenbozzz