groupgets/pylepton #52: OSError: [Errno 90] Message too long on Raspberry Pi4

BatbayarU, 2021-01-16:
Hi everyone,

I'm implementing a project using Lepton 2.5 with breakout board v2.0 on Raspberry Pi4. About a few months ago, it was working well. But now it isn't working, and an error below occurs.

> Traceback (most recent call last):
>   File "/home/pi/Desktop/lepton test.py", line 7, in <module>
>     a,_ = l.capture()
>   File "/usr/local/lib/python3.7/dist-packages/pylepton/Lepton.py", line 153, in capture
>     Lepton.capture_segment(self.__handle, self.__xmit_buf, self.__msg_size, self.__capture_buf[0])
>   File "/usr/local/lib/python3.7/dist-packages/pylepton/Lepton.py", line 125, in capture_segment
>     ret = ioctl(handle, iow, xs_buf[xs_size * (60 - messages):], True)
> OSError: [Errno 90] Message too long

Do you know what is wrong with it?

Thank you very much.
Batbayar

Oroelipas, 2021-01-28:
I have the same error.   
raspberrypi_video from LaptonModule is also working for me but not this library.
I am using Lpeton 3.5 with breakoutboard v2.0 on Raspberry Pi 4.

OS: Raspbian Buster
Kernel: 5.4.72-v7l+ 
Python: 3.7.3 

Any solution?

Oroelipas, 2021-02-01:
**SOLUTION:**

The problem is in the SPI buffer size. The default value is 4096 but 160x120 images from Lepton3 are bigger. 

You need to increase the size of the SPI buffer as explain [here](https://stackoverflow.com/questions/16427996/increase-spi-buffer-size-in-raspbian/16440226)

Edit `/boot/cmdline.txt` and add in the end:
`spidev.bufsiz= 65535`

Then reboot your raspberry and check it worked with:
`cat /sys/module/spidev/parameters/bufsiz`

Hope it works for you also. Maybe this information should be in the README setup tutorial

BatbayarU, 2021-02-02:
> **SOLUTION:**
>
> The problem is in the SPI buffer size. The default value is 4096 but 160x120 images from Lepton3 are bigger.
>
> You need to increase the size of the SPI buffer as explain [...]

Hi,
Thank you for your solution. I have tried this but the I couldn't change the buffer size. However, I solved this the opposite of your solution which is by reducing the message size of the library. SPIDEV_MESSAGE_LIMIT = 24 to 8 of Lepton.py. I hope this helps someone.

closed 2021-02-02 by BatbayarU