groupgets/pylepton #54: Lepton3/3.5 using pylepton

TinManAkshay, 2021-02-09:
Hi,

Does anybody try to get lepton 3 or 3.5 successfully working with these libraries. What changes in the libraries could be required to get lepton 3 working? 

We are trying to switch from lepton 2.5 to 3/3.5 for better thermal imaging.

Please let me know.

kekiefer, 2021-02-13:
You should check out the code on the `lepton3` or `lepton3-dev` branches that wil lwork with the lepton 3/3.5

closed 2021-02-13 by kekiefer

TinManAkshay, 2021-02-16:
Hi @kekiefer, 

I wanted to ask a quick question. Have you ever got lepton 3 or 3.5 fully working with these branches lepton3 or lepton3-dev using Breakout board v2.0? I'd appreciate the information.

Thanks,
Akshay

kekiefer, 2021-02-16:
Yes. Both 3.0 and 3.5 worked with this code at least at the time those were developed.

TinManAkshay, 2021-02-17:
Hi @kekiefer,

Can you please tell me the reason behind the **frequent thermal camera shuttering** as I believe it could be because of sync with the SPI protocol between Lepton and Pi? 

Thanks,
Akshay

Tylersuard, 2021-09-17:
@TinManAkshay I have that same thing

TinManAkshay, 2022-03-24:
Hi @Tylersuard  - 

Have you ever got lepton 3 or 3.5 working properly with pylepton libraries with no issues?

Tylersuard, 2022-03-24:
@TinManAkshay no I did not, I think I just ended up using a 3rd party software, tCam mini.  https://groupgets.com/manufacturers/danjuliodesigns-llc/products/tcam-mini