groupgets/pylepton #55: RE: pylepton and error issues w/ Python3, BeagleBone Green, and LCD Display

silver2row, 2021-02-13:
Hello,

I tried the sudo python3 setup.py install command after sudo apt install python3-numpy python3-opencv.

It seems the script built the installs but I am still receiving issues/errors. 

...

Anyway, as I make some headway on this nice SDK on Python(3) w/ changing up some source, is /var/mail/pylepton a real dir. to have my pylepton SDK located in?

Seth

P.S. It seems this is one of the many errors so far.

    ./LeptonOne.py
    ./LeptonOne.py: line 1: import: command not found
    ./LeptonOne.py: line 2: import: command not found
    from: can't read /var/mail/pylepton
    ./LeptonOne.py: line 5: syntax error near unexpected token `('

Where the source is:

    import numpy as np
    import cv2
    from pylepton import Lepton

    with Lepton() as l:
      a,_ = l.capture()

    cv2.normalize(a, a, 0, 65535, cv2.NORM_MINMAX) # extend contrast
    np.right_shift(a, 8, a) # fit data into 8 bits
    cv2.imwrite("FirstPhoto.jpg", np.uint8(a)) # write it!

If you are understanding my issue before I find it, please let me know.

silver2row, 2021-02-13:
Hello,

Forget it. I configured the .py file for the lib. and SDK to use python3 correctly this time.

Seth

P.S. I was trying w/out understanding.

silver2row, 2021-02-13:
Hello,

Just to reopen the issue for a bit. What would be the cause for this photo?

https://imgur.com/XRUcWsm

I updated the scripts to run as Python3 and thought it would work as it is always just not easy enough. Anyway, I tried. The photo is odd but it is supposed to be my hand.

Seth

P.S. That is a photo taken on a LCD Screen that is attached to the BeagleBone Green.

kekiefer, 2021-02-13:
You want to switch over the the branch `lepton3` or `lepton3-dev`. The main branch of the code only supports lepton 2.

silver2row, 2021-02-13:
Hello,

I have a lepton v2 module. https://www.digikey.com/short/4342hc is my camera and this is the Lepton v2 module: https://www.digikey.com/short/4342h5.

Do these still require me to use lepton 3?

Seth

closed 2021-02-16 by silver2row