groupgets/pylepton #56: Shutter closes and opens

TinManAkshay, 2021-02-25:
Hi,

I've been successful with Lepton 2.5 w/ Breakout board v2.0 and Pi 3 B+ using these libraries. However, when I run the same thing w/ Pi4, I get frequent shutter closes and opens on power on. Its about every 3-5 seconds. Additionally, data coming out of my other sensors connected on I2C protocol freeze and come back alive and then repeat. Although, lepton 2.5 never worked properly.

I don't know what's this behavior when I switched to Pi 4. Has anyone gone through this situation or have any clues about the root cause and potential methods to fix the problem.

Thanks,
Akshay

kekiefer, 2021-02-25:
I don't know about your FFC issues, but regarding differences between older raspbian releases and those that support pi4, there was some frequency scaling that can affect SPI transfer rates. To stop the frequency scaling, you can try the following:
```
sudo sh -c "echo performance > /sys/devices/system/cpu/cpufreq/policy0/scaling_governor"
```

TinManAkshay, 2021-02-25:
Hi,

Thank you for that suggestion and I did make that change.

1. I installed cpufrequtils and I changed GOVERNOR = performance as a default option. Now the frequency has been changed from 700Mhz to 1.5Ghz. I believe there is no issue on that.
2. Also, I have increased the SPI buffer size to 65536.
3. I changed SPI_MESSAGE_LIMIT = 59 for better spi reading. I hope that does not create issues too.

What's your feedback on three changes I made as stated above?

Thanks,
Akshay

closed 2021-03-11 by TinManAkshay