groupgets/pylepton #57: lepton 3.5 not working with RPi0W

vidmo91, 2021-03-12:
Hi! 
I cannot get Lepton 3.5 with Breakout Board v2.0 working on RPi0W. I get horizontal bands as in attached image.
I increased SPI buffer, so SPIDEV_MESSAGE_LIMIT = 24 works fine I tried 48 and 8 with same result.
The same setup works fine on RPi4.
Any ideas how to solve that?
[image]

Retaehc-pop, 2021-04-07:
i have meet tis error as well. have you finally found the solution yet?
[image]

anted19, 2023-08-18:
tengo el mismo problema , nadie lo soluciono ?

anted19, 2023-08-19:
Por fin te go la solucion , cambia en el codigo de Lepton.py : SPEED=20000000 MHZ , la velocidad del SPI es muy lento

anted19, 2023-08-19:
/
[image]