groupgets/pylepton #58: Error: Inappropriate ioctl for device

AravindSadashiv1998, 2021-04-20:
Hi,
I'm using the FLIR lepton 3.5 with the breakoutboard v2.0 and when I'm running the command "sudo pylepton_capture output.png", I'm getting the error " Error: Inappropriate ioctl for device ". Can anyone please help me with this as it's urgent. 

Also what is the pinlayout specified?

Best,
Aravind

OwenTC, 2021-06-14:
I am having the same problem right now using it through USB. Did you ever figure out a solution to the error?

Thanks,
Owen

mishave, 2021-06-18:
@AravindSadashiv1998 just to what your output is when you run:
```cat /sys/module/spidev/parameters/bufsiz ```

Typical this should typically return 4096 bytes by default.. however when messing around trying to increase this number to the suggested 65535, the buffer returning an error or a very low number i.e. 4 when check with the above cmd. this in turn creates the 'Inappropriate ioctl for device '
to fix it I went ahead and reduced that size down using:
```sudo nano /boot/cmdline.txt``` and add in the end:
```spidev.bufsiz= 131072```
Reboot after and check again.

Tylersuard, 2021-09-12:
@mishave I just tried that, still the same error.

jcarolinares, 2021-10-06:
Hello, 

I'm experimenting the same error, in my case I've tried with a Raspberry Pi 4 model B the error message is the same but cat /sys/module/spidev/parameters/bufsiz does not exist

Any idea about what is happening or what are we doing wrong?

Thanks!

mishave, 2021-11-03:
> @mishave I just tried that, still the same error.

Apologies, just saw this, will set one up over the next week and see what comes up, I do remember it being a pain to set up. from memory, I did it using the lepton3-dev branch i.e.`git clone -b lepton3-dev https://github.com/groupgets/pylepton.git` keep in mind there are a few things you need to fix within that.

mishave, 2021-11-04:
@Tylersuard @jcarolinares Alrighty so, i started with a fresh build so had to reinstall everything first up opencv install issues-
`pip3 install opencv-python` > would not work getting such on install, so I tried installing the previous build as per below:
`pip3 install opencv-python==4.5.3.56`
also installed NumPy:
`pip install -U numpy'

once that was installed I made a copy of the lepton3-dev branch:
`git clone -b lepton3-dev https://github.com/groupgets/pylepton.git`

ran install `sudo python setup.py install` then tried to run `pylepton_capture output.png` which threw a cv2 import error..

Tried installing the contrib package and manually some of its dependencies using:
`pip3 install opencv-contrib-python; sudo apt-get install -y libatlas-base-dev libhdf5-dev libhdf5-serial-dev libatlas-base-dev libjasper-dev  libqtgui4  libqt4-test`

after that I was getting the ioctl error:
pi@raspberrypi:~/pylepton $ pylepton_capture output.png
Traceback (most recent call last):
  File "/usr/local/bin/pylepton_capture", line 37, in <module>
    image = capture(flip_v = options.flip_v, device = options.device)
  File "/usr/local/bin/pylepton_capture", line 10, in capture
    a,_ = l.capture()
  File "/usr/local/lib/python3.7/dist-packages/pylepton/Lepton3.py", line 55, in capture
    Lepton.capture_segment(self._handle, self._xmit_buf[:self._msg_size * Lepton.ROWS], self._msg_size, self._capture_buf[0])
  File "/usr/local/lib/python3.7/dist-packages/pylepton/Lepton.py", line 125, in capture_segment
    ret = ioctl(handle, iow, xs_buf[xs_size * (60 - messages):], True)
OSError: [Errno 90] Message too long

So I checked the buf size and it returned:
pi@raspberrypi:~/pylepton $ cat /sys/module/spidev/parameters/bufsiz
4096

Extended the buf size:
pi@raspberrypi:~/pylepton $ sudo nano /boot/cmdline.txt
and add the following to the end:
spidev.bufsiz=65535

pi@raspberrypi:~/pylepton $ sudo reboot

Run the in built script
sudo pylepton_capture output.png

happy days.

kekiefer, 2022-01-26:
#66 merged which reduces the spidev buffer size again, because apparently it works with the smaller message sizes. Let me know if there are issues with this approach.

closed 2022-01-26 by kekiefer