groupgets/pylepton #65: FLIR Lepton 3.5 with PyLepton in Jetson Nano

PhamHoangBao, 2021-11-28:
Hi everyone,
I currently run Pylepton with Lepton Breakout Board V2 with FLIR Lepton 3.5 in Jetson Nano. I used branch Lepton3 and got the error "Garbage frame number resetting ..." continuously. 
Is there anyone have run Pylepton in Jetson Nano can give me recommendation about it ? The Pylepton have worked perfectly on my rasberry Pi4 .
Thank you so much.

AlexandrTurok, 2022-01-05:
Hello.
Is there a solution to this issue?

sethuageye, 2022-01-06:
Working with Jetson Nano and Lepton 3.5.
Used the "lepton3-dev" branch of this repo in Raspberry Pi 4 and was able to succesfully capture images.
But the same repo did not work in Jetson Nano (got the same error "Garbage frame number resetting ..." continuously.)

I tried using another C++ based repo for interfacting Lepton 3.5 with Jetson Nano _https://github.com/Myzhar/Lepton3_Jetson_

The above repo is successfully able to stream thermal data in Jetson Nano. But, the accuracy of the radiometric data obtained using this repo does not reflect real-time temperature values (throughout the range of Lepton 3.5, both in High & Low gain mode)

_**But, pylepton** offers better radiometric accuracy throught the range of the sensor (Tested using Rpi4).**

**So, need to get pylepton working in Jetson Nano**_

**My observation:**
The error message "Garbage frame number resetting ..."  is due to SPI communication failure. 
SPI has been enabled and working seamlessly in Jetson Nano (because the C++ library is able to successfully stream data)

**So, the SPI configuration made in pylepton does not seem to match for Jetson Nano**

If anyone has come across the solution, kindly suggest. 
Thanks in advance!!

acqxi, 2022-08-06:
Hello.
Is there a solution to this issue?

> 

Hello.
Is there a solution to this issue?

XillanChang, 2022-08-08:
Hello.
I changed spi buffer size to 65536, and also changed spidev message limit to 8.
But it didn't work and had same question =>OSError: [Errno 90] Message too long

yasin-aydn, 2024-03-15:
Hello.
Is there a solution to this issue?

XillanChang, 2024-03-15:
I give up that runing Pylepton with Lepton Breakout Board V2 with FLIR Lepton 3.5 in Jetson Nano.
So I use PureThermal 2 I/O Board with FLIR Lepton 3.5 in Jetson Nano.

Lucas8377, 2024-12-24:
Hi was it available for Lepton 3.5 on PureThermal 2 Board

Lucas8377, 2025-01-02:
In my opinion, PureThermal Board is more convenient for developing the project on NANO.
You could try it.