groupgets/pylepton #69: Lepton3 - Module is not callable

skhatri15, 2022-04-15:
I am trying to use Lepton 3.5, along with python. I am trying to use Pylepton library with my python3 and Raspberrypi 4 Model B.

I just found that there has been a branch updated about 3 months ago for a new file called Lepton3.py. I am trying to import this file in my code, however I am getting this when trying to import: "Module is not callable" from Pyright reportGeneralTypeIssues.

My code: 
=============================================================

import sys
import numpy as np
from cv2 import cv2
from pylepton import Lepton3

def capturePic():
    **with Lepton3() as l:**
        a,_ = l.capture()
    cv2.normalize(a, a, 0, 65535, cv2.NORM_MINMAX)
    np.right_shift(a, 8, a) #fits the data into 8 bits)
    cv2.imwrite("output.jpg", np.uint8(a)) #writes image to file

if __name__ == '__main__':

    print("capturing image and saving to output.jpg in current directory")
    capturePic()

=============================================================

In the **"def capturePic():"** function, I have changed the **"with Lepton3() as l:**" to **"with Lepton(): as l:"** and it still does not work and the error becomes   **" 'Lepton' is not defined "**

I looked and compared the Lepton.py file and Lepton3.py file, and the only difference in the class I see is Lepton.py has class Lepton(object): and Lepton3.py has class Lepton3(Lepton): as its definition.

I tried using **"with Lepton3(Lepton) as l:"** but that also did not work.

Could someone assist with this issue if you have experienced/resolved this on your end?

robots, 2022-04-19:
change:
with Lepton3() as l:

to:
with Lepton3 as l:

i believe there is error :-)

Antuto, 2022-05-25:
Hello !
Simply change the `from pylepton import Lepton3` to `from pylepton.Lepton3 import Lepton3` and you should work ;)

AbhishekGupta2002, 2023-09-13:
its not working the same error is still coming